nefarius/ScpToolkit #377: Not working with other apps and input problems

TheDarkness73, 2016-07-12:
So I was wondering if I could get this to work with input mapper, because currently if I start up input mapper it doesn't recognize this controller

The other this is it seems sometimes my input will be continously held like if I turn it'll keep turning. I'm assuming that's just a latency issue or problem with bluetooth, correct?

merklaw, 2016-07-21:
Please tell us what version of the ScpToolkin Driver installer you used (found in the top of the Log) and what is the full name of the "input mapper" application.

About your other issue, what application or game are you using/playing when this problem happens?

TheDarkness73, 2016-07-22:
It was mainly fallout for the disconnect, but that seems to be working just fine as of lately.

The application name is InputMapper  on version 1.5.21.19358. 

My version of SCP I am using is ScpToolkit v1.6.238.16010

merklaw, 2016-07-30:
This issues seems to be resolved.

Jan200101, 2016-08-17:
please close issues if they are resolved

closed 2018-01-26 by nefarius