nefarius/ScpToolkit #378: SCP - not installing.

subzerocro, 2016-07-12:
Ever since i tried the new version of scp, it fucked up all my drivers and i had to spend 2 hours fixing my mouse. After i got everything working again i relised that my controller wasnt working, I also bought a new PC controller but niether does it work. I tried installing SCP drivers again they only installs BUS Drivers and DS3 usb drivers and doesnt install SCP DS3 Service. Is there any way to fix it, I kinda want to use my controller and not waste money because of scp

nefarius, 2016-07-16:
Since you didn't include any information on which version you used I assume you haven't tried the latest 1.7 beta?

merklaw, 2016-07-21:
To: subzerocro
Please tell us whether your mouse is a Bluetooth mouse, and what you mean by a "PC controller", as that name is too generic to be of use. Also, did you use the Uninstall functionality included with the ScpToolkit Setup application?

closed 2018-01-26 by nefarius