nefarius/ScpToolkit #380: Make driver compatible again with arbitrary PIDs

mirh, 2016-07-16:
I see how funnily I was one of those [praying](https://github.com/nefarius/ScpToolkit/issues/116) for WinUSB to come back.. 
But now I have this laptop with built-in BT which has an unsupported PID (it's still 100% an Atheros AR3012.. just with Asus's VID instead of Qualcomm's) and the 1.7 beta didn't work. 

Is "complain to dev and wait newer strings to be included" the only approach, could #290 improve this, or what else?

nefarius, 2016-07-16:
Holy sh*** you are the master of GitHub, I can't read this fast, gimme a few minutes =D

mirh, 2016-07-16:
Oh, I'm just underlining the existing problem d:
And, if not #290 (would be interested to know about) at least it's going to be a kind of  [META] for every bug regarded to missing PIDs.

nefarius, 2016-07-16:
I'm slowly catching up; my mail server died the day I went on vacation so I currently can't get interrupted by anyone xD

DragonWolf5589, 2016-08-22:
does anyone know if this is resolved yet?

bluetooth worked fine in the latest release, but cpu usage was 60-80%! 

the lastest beta works perfect but shows I do not have a bluetooth connection (I reinstalled the orginal driver) I dont use a dongle its built in, and ive NEVER used it except for the ps3 controller anyway .

how do I install it as the button wont "light up" for me on the installer

(and some reason my ps3 controller behaves badly when wired for some reason, it only works correctly with bluetooth)

up to now its either random spasmic behaviour, or 90-100% total cpu usage of with over 60% is from SCPServer!

mirh, 2016-08-23:
> does anyone know if this is resolved yet?

_[We are getting there](https://twitter.com/CNefarius/status/763447847337484288)_.

closed 2018-01-26 by nefarius