nefarius/ScpToolkit #382: Installing driver makes BT dongle stop working

TheWingedCherryPie, 2016-07-17:
(win 10 x64)
So I go to scp driver installer and install the bluetooth dongle driver for the dongle I'm using, and when it installs, the dongle stops working, and scp service can't request bluetooth hosting because the dongle doesn't work anymore. When I uninstall the driver, the dongle starts working again, but it won't connect to the DS3 controller because the driver isn't available.

merklaw, 2016-07-21:
That is expected behavior. Mine does it as well. I have a Bluetooth speaker, and whenever I want to use it, I have to uninstall the SCP driver to my speaker to work. You are welcome to contribute patches to resolve the problem.

nefarius, 2016-07-21:
Since this is expected behavior it's not worth keeping an open issue.

closed 2016-07-21 by nefarius