nefarius/ScpToolkit #383: my Joystick disconnected

herym15, 2016-07-18:
Please Help!!!
My controller get disconnect when the vibration's on, but with old scp driver version (original) it's working. Both the vibration and the controller work perfect.
my controller is ps 2 controller with converter to ps 3 controller, so it's detected as play station 3 ® Controller.

merklaw, 2016-07-21:
If you would like to contribute to the resolution of the problem, try to find out exactly on what commit it stopped working. It will help the developers identify and fix the problem.

herym15, 2016-07-23:
Its a converter to ps 3 controller by using ps 2 controller
here is the pic>>>>>
https://cloud.githubusercontent.com/assets/20520651/17077306/9aa3a71e-50f5-11e6-9a87-c39d1a88833a.jpg
how can i find its product ID??

closed 2018-01-26 by nefarius