nefarius/ScpToolkit #384: Driver Installer Will Not Run

darianb1031, 2016-07-22:
I installed the toolkit, and when I went to run the driver installer, nothing happened. I opened up Task Manager and tried it again, and then I saw that there is a brief moment of a few seconds where the driver installer task appears, but then it disappears and nothing happens. There's no error message or anything associated with it, it just doesn't run. 

In addition, the tray app also crashes if I try to interact with it in any way. 

I've tried uninstalling and reinstalling, and repairing from the installer.

merklaw, 2016-07-22:
It is very hard to diagnose the problem without detailed information or error codes. Can you take a look at the folder "C:\Program Files\Nefarius Software Solutions\ScpToolkit\Logs" and see if any of the logs mention anything wrong?

darianb1031, 2016-07-22:
There's one log file - an xml- that keeps getting overwritten every time I attempt to open the driver installer. None of the xml programs I've tried have been able to open it properly because they all have a problem with the log4j prefix. I got it to open in Wordpad and the document is very long and I'm not sure what I'm looking for to determine if there was a problem.

merklaw, 2016-07-30:
Are you running the latest version of ScpToolkit? If so, would you mind uploading the log files somewhere, like pastebin?

closed 2018-01-26 by nefarius