nefarius/ScpToolkit #385: Only slots 1/2 recognized in-game, all connect fine

SqMonkey87, 2016-07-23:
After finally springing for one of the dongles off the supported list (the asus one, I believe) all my contollers (3 PS3, 1 PS4, all genuine) connect via USB and bluetooth like they should.  However, while in-game, any game, I only get response from controller slots 1/2.

Tried moving controllers around in monitor to check controllers, all work when in slots 1 or 2.  Have uninstalled/reinstalled a couple times to same result.

Not sure what my next step should be... any suggestions?

charlesmilette, 2016-07-23:
Maybe you need to have a game with 4 players multiplayer instead than 2 to have it recognize the other controllers. Also notice that XInput support only up to 4 controllers.

To make sure the system recognizes the other controllers, you can also [list the connected controllers](http://www.addiehf.com/2016/03/29/how-to-calibrate-your-gaming-controller-in-windows-10/).

For example, there is only one controller connected in this computer:

[image]

SqMonkey87, 2016-07-24:
Right, that's already the case.  First the plan was 4 player split-screen rocket league but when only 2 slots would go I thought maybe pic version couldn't do that so we tried towerfall and samurai gunn, etc.  we do a lot of different 4 player local games, so lots to choose from.  The issue was all controllers say connected but not being recognized in-game.

I will look into the system menu when I get home, I've just been going off them connected to scp which you can see in scp monitor, and since you have to choose the connected controllers when installing the scp controller drivers I assumed the system recognized them already, you know?

Thanks for the suggestion though, it for some reason I find the problem through the system settings I'll update the post.

SqMonkey87, 2016-07-24:
I suppose I should add that I only those 4 controllers have been used on my pc, so if xinput works such that having used other controllers previously meant new ones couldn't have a profile it wouldn't be the case in this particular instance.

closed 2018-01-26 by nefarius