nefarius/ScpToolkit #386: Dragon War PS3 Controller support

pepijnijben, 2016-07-24:
Hi I have a PS3 Controller from dragon war and was wondring if you could add support for this controller. Below you will find a link to this controller and if I can help in anyway pls let me know.

http://www.dragonwar.eu/product-detail.php?id=26

merklaw, 2016-07-30:
Please post the Hardware ID of the controller. You can find your Hardware ID in Windows Device Manager if you right click on the device and click Properties, and then go to the Details tab and find "Hardware IDs". Please note if there are other entries that have something with "ID" in them, and post them here as well.

pepijnijben, 2016-07-30:
Here are all the IDs I found for the controller.

Hardware Ids:
```
USB\VID_054C&PID_0268&REV_0100
USB\VID_054C&PID_0268
```

Compatible Ids:
```
USB\Class_03&SubClass_00&Prot_00
USB\Class_03&SubClass_00
USB\Class_03
```

Matching device Id:
`USB\VID_054C&PID_0268`

Base container Id:
`{39a3769e-45aa-11e6-9db0-001a7dda7113}`

Configuration Id:
`oem30.inf:USB\VID_054C&PID_0268,USB_Install`

Container Id:
`{39a3769e-45aa-11e6-9db0-001a7dda7113}`

Reported device Ids hash:
`F949201D`

closed 2018-01-26 by nefarius