nefarius/ScpToolkit #387: Device Dectected but Not Working

Skyferia, 2016-07-25:
Issue as in image below:
[image]

Lights @ controller flashes.

Also the SCP Monitor status:
[image]

What I installed:
[image]

Also re-installed several times, ran_ScpToolkit Clean Wipe Utility_ prior to each install.

Controller is original PS3 DS3.

Please advice. Thank you.

Skyferia, 2016-07-26:
Never mind, fixed it.

How: Reset controller
How to reset: Used a toothpick and pressed @ the small button in the small hole @ the back of controller

Hope this will help others.

tosho, 2016-08-02:
Reseting the controller doesn't work for me (PS3 DS3 / Windows 10 x64). 
I tried to uninstall my bluetooth drivers and SCPToolkit and restarting a lot but the GamePad can't be found in the games and in the Control Panel > Game Controllers. SCP Monitor shows the controller to be connected (both with cabel and via bluetooth) also I can do Input Tester which shows that the controller is working but still can't be found in all my games. 
This happened just today. I had previous problem with the Bluetooth - when I restart the PC the controller can't connect via Bluetooth so I decided to Uninstall all and make clean install. I haven't be so wrong.

closed 2018-01-26 by nefarius