nefarius/ScpToolkit #390: fake PlayStation 3 controller vibration  via bluetooth fix

7710101, 2016-07-27:
https://drive.google.com/file/d/0B02BQcVF2DX9ajc1NTdfRnRpNkU/view?usp=sharing

merklaw, 2016-07-30:
Please add more details.

cesc007, 2016-07-30:
it works with windows 10, thanks a lot!

n00bC0d3, 2016-08-02:
not work with mine ...
using 
intel bluetooth 4.0  VID 8087 PID 07DC

with ScpToolkit v1.6.238.16010 

unable to pair

7710101, 2016-08-02:
You must use USB dongle 2.0 to 4.0 built-in Bluetooth will reinstall the driver's on restart

7710101, 2016-08-02:
I have a HP envy laptop and i have the same problem. When i got my Bluetooth dongle it works with the (fake ps3) controller and my Bluetooth headphone works great with the intel Bluetooth 4.0

n00bC0d3, 2016-08-04:
my bluetooth is internal bluetooth in lenovo y50-70 touch. 
it is bluetooth 4.0 , so it should be compatible right ?

7710101, 2016-08-04:
Built-in bluetooth will not work. U must get a usb dongle for this to work

n00bC0d3, 2016-08-05:
[image]

what i done : 
1. stop replaced the SCP DSx service 
2. replace scpcontroll.dll with fakeFix 
3. start the service back ..

it's the same whether i use version 1.6 or 1.7

7710101, 2016-08-05:
did u get the usb bluetooth dongle  its the only whey it will work
internal bluetooth will not work windows is built to reinstall removed dirvers

n00bC0d3, 2016-08-06:
no, i still trying to figure how the trick to use built in bluetooth. 
although i didn't restart windows after updating all driver to SCPTool driver, still can't pair . 
Bluetooth status always NONE. 

wonder if anyone can give me clue about "bluetooth pairing status NONE "

Freeedo, 2017-12-23:
Download the file and then what?  "first intall scpserver then move and replace scpcontrol.dll in bin folder". I get "fatal error" when I try to start up scp again.  Can you at least give us a step by step instead of just throwing a file up and saying "restart hope for the best...".

7710101, 2017-12-23:
[Scarlet.Crush Productions.zip](https://github.com/nefarius/ScpToolkit/files/1584468/Scarlet.Crush.Productions.zip)
extract the file directly to the (C drive) and then install if it doesn't work try force install or (disable driver signification) restart and hope for the best.

closed 2018-01-26 by nefarius