nefarius/ScpToolkit #391: Usage of internal Bluetooth

acocalypso, 2016-07-27:
Is it possible to add the usage of a build in Bluetoothmodule?

F.e im using my Dell venue 8 Pro with Windows 10 and I do only own 1 micro USB Port... thats mostly used for charging.

So it would be great if I could instead convert my internal bluetooth to make it work with my controller.

merklaw, 2016-07-30:
You can find your Hardware ID of the Bluetooth adapter in Windows Device Manager if you right click on the device (found under "Bluetooth Radios") and click Properties, and then go to the Details tab and find "Hardware IDs". Please note if there are other entries that have something with "ID" in them, and post them here as well.

mirh, 2016-08-05:
#380

closed 2018-01-26 by nefarius