nefarius/ScpToolkit #394: Bluetooth Pairing Issue

tomose, 2016-07-30:
Win7 x64
Sony DS4 controller
Bluetooth controller VID: 05AC PID8218

Without ScpToolkit, windows can pair the controller just fine. Obviously, I get no functionality out of this, BUT a pairing is established if I tell it to add devices then hit Share+PS button on controller.

With ScpToolkit installed, I go to Add a device, put the controller into pairing mode, and it never shows up.

Back to the desktop, if I only press the PS button on the controller, ScpToolkit tray notifications tells me that a Sony Dualshock 4 controller is found, BUT no pairing is done and the device does not show up in Devices and Printers.

So what is the proper way to pair this controller using ScpToolkit + Bluetooth + Win7? Should what I be doing be working or am I missing some step here? I have uninstalled and reinstalled the drivers in different combinations and restarted many times already.

tomose, 2016-07-30:
I should mention that connection via USB cable works with ScpToolkit, but I would certainly like to get BT support working here.

merklaw, 2016-07-30:
Please post your Bluetooth adapter Hardware ID. You can find your Hardware ID of the Bluetooth adapter in Windows Device Manager if you right click on the device (found under "Bluetooth Radios") and click Properties, and then go to the Details tab and find "Hardware IDs". Please note if there are other entries that have something with "ID" in them, and post them here as well.

tomose, 2016-07-31:
I believe I did already in my original post. Third line: Bluetooth controller VID: 05AC PID8218

Was that not the format or proper ID?

The only additional info in device manager for hardware ID for the BT adapter is REV_0042

mirh, 2016-08-05:
#380

closed 2018-01-26 by nefarius