nefarius/ScpToolkit #396: Bluetooth Pairing

protectivedad, 2016-08-02:
My Bluetooth ID is:
USB\VID_413C&PID_8160&REV_0173
USB\VID_413C&PID_8160

The libusbK device driver is loaded. 

This SCP Monitor Legacy shows:
[image]
when the controller is plugged into the USB. When I unplug the USB and try the controller from the BT I get a message from the tray app saying Geniune Sony DualShock 3 found. But it won't pair. It worked once but when I restarted the pc it stopped pairing. 

I've included the debug zip also.
[ScpToolkit_Log-Package_antho_1470176303.zip](https://github.com/nefarius/ScpToolkit/files/398293/ScpToolkit_Log-Package_antho_1470176303.zip)

s0me-1, 2016-08-05:
Did you tried to force the pairing using the old legace pair utility ? 

[image]

( the address to be setted must be the host adress )

mirh, 2016-08-05:
#380

mirh, 2016-08-05:
Your adapter isn't included in driver .inf. 
You can wait for newer GLORIOUS version to come out, or either downgrade to 1.6 or old SCP release.

protectivedad, 2016-08-05:
I'm using version 1.6 I tried as low as 1.6.2, but I get the same problem. I could never get the old SCP release to work. Only SCPToolkit and only until I did a reboot.

mirh, 2016-08-05:
Uhh, then your problem is [this](https://github.com/nefarius/ScpToolkit/issues/224#issuecomment-192291387).

OverWorldD, 2017-04-04:
I have this problem with one of my controllers. Both are clones, One is a bit better and works as it should even over BT, the other one works only over USB, and gets detected as a genuine DS3 controller (It's most certainly not), but does not seem to pair okay. Attachment shows what my server does, it goes as far as this when I start the controller, and then the controller shuts down after a while.
[image]
It works fine with sixaxis controller to my android tablet over bluetooth so I don't know why SCP doesn't like it :(

closed 2018-01-26 by nefarius