nefarius/ScpToolkit #398: Bluetooth host adress is corrupted after reboot

s0me-1, 2016-08-05:
I had the case with the old 1.2.2 SCP aswell, When I do a clean install, all the drivers are correctily installed and everything is working great. But if I reboot my computer, i have this : "Host adress : <Error>"

Here's the attached log created by your bug report tool :)

[ScpToolkit_Log-Package_arnau_1470400140.zip](https://github.com/nefarius/ScpToolkit/files/403733/ScpToolkit_Log-Package_arnau_1470400140.zip)

charlesmilette, 2016-08-05:
That could be related to issue #51

s0me-1, 2016-08-05:
Indeed, I have an Internal bluetooth chipset too, regarding the answer from Nefarius, it seems the old Uninstall/Reinstall is the only workaround to get it working :p

mirh, 2016-08-23:
https://github.com/nefarius/ScpToolkit/issues/224#issuecomment-192291387

closed 2018-01-26 by nefarius