nefarius/ScpToolkit #399: Exclusive Mode Broken In Latest Windows 10 Update (Aug 3rd)

Teutonic84, 2016-08-05:
So exclusive mode is once again broken in the latest update to Windows 10 that happened on August 3rd. Inputmapper has released a tool that fixes the exclusive mode issue (https://inputmapper.com/downloads/download/5-tools/63-duleshock-4-exclusive-mode-tool). Are there any plans to fix this in ScpToolkit as well?

nefarius, 2016-08-05:
I wasn't aware the toolkit is affected? To my knowledge it isn't.

charlesmilette, 2016-08-05:
It's working perfectly fine on the Anniversary Update for me.

Paranoid-Andr01d, 2016-12-24:
The "PS" button no longer works for the Dualshock 3 after the Anniversary Update for me. I believe this is due to lack of exclusive mode. All other buttons are fine but I like having a "Home" button for Steam Big Picture.

closed 2016-12-30 by nefarius