nefarius/ScpToolkit #401: ScpToolkit Profile Manager - Not working/saving

Controlzurg, 2016-08-06:
Pretty much ScpToolkit Profile Manager is not saving or changing the buttons when I save it.. 
I know it's in pre-alpha but anyway thought I'd say as I thought it would work..

Anyway,

Thanks

nefarius, 2016-08-10:
It simply is a piece of sh*t, sorry for that :laughing:

Stealthmate, 2016-08-20:
This needs to be implemented. The way it currently is, games recognize the controller as an official xbox360 one, and sometimes assign mappings (fuck you Ubisoft) which simply suck. It doesn't have to be anything fancy, but I think it would be awesome if we could at least remap the controller itself (e.g. swap R2/L2 with R1/L2). Afaik this was doable in the original scp monitor.

Either this, or make it so that it doesn't pretend to be an official controller so that we don't fall prey to stupid company decisions (although I'm not sure if this wouldn't introduce other issues with other games).

blar210, 2016-08-28:
Ya I agree with Stealthmate. I wanted to make a profile to switch betweeen Xbox and PS3 configs (switching the triggers with shoulders and shoulders with triggers) but I ran into the problem of R2 mapping to R1 and L2 mapping to L1 but the mappings of L1 and R1 would not register so I was left with two sets R1 and L1.

charlesmilette, 2016-08-29:
Look at the bright side!

You can now shoot at twice the rate you usually do!

blar210, 2016-08-29:
I wish, but on pc to fire is usally the triggers but if i try to switch them i only have shoulder function buttons.

closed 2018-01-26 by nefarius