nefarius/ScpToolkit #405: Get Mayflash PS3003 adapter to work with ScpToolkit

utjduo, 2016-08-08:
Quick background info:
The Mayflash PS3003 is a popular adapter used my a lot of users of the Steam Link. It lets you connect up to 4 PS3 controllers through the Steam Link to your computer (this is the only way to do this with the PS3 controllers, that why it's popular). The only problem is that the adapter doesn't work with ScpToolkit and this is why:

The way the Mayflash adapter works is that it creates 4 "Human Interface Devices" that is the controllers. These controllers have these Hardware IDs:
* HID\VID_0079&PID_1801&REV_0100&Col04
* HID\VID_0079&PID_1801&REV_0100&Col03
* HID\VID_0079&PID_1801&REV_0100&Col02
* HID\VID_0079&PID_1801&REV_0100&Col01
They simply forward the commands from the PS3 controllers, so it would be the same as plugging the controller into the computer itself.
Because this is now Mayflash ID's ScpToolkit doesn't recognize them as PS3 controllers.
I'm guessing all that has to be done is to add these hardware IDs to the list of controllers supported in ScpToolkit and place them as normal PS3 controllers so when installing the drivers for the Mayflash controllers it would be the same as installing drivers for the real PS3 controllers plugged into the computer.

I'm completely new to this code and haven't found where to add this but I know it's possible. I helped Scarlet.Crush add support for fake PS3 controllers back when he was still developing by giving him the hardware IDs and later beta-test.
I hope someone here can help me get this support for this great software.

Thanks!

utjduo, 2016-08-13:
UPDATE! I found the Gamepad Analyzer in the latest beta and analyzed my ps3 controller through Mayflash 3003.
[Mayflash PS3 Game Controller Adapter_hid-report.dump.txt](https://github.com/nefarius/ScpToolkit/files/416526/Mayflash.PS3.Game.Controller.Adapter_hid-report.dump.txt)

I did this for controller nr1 but there are up to 4 controllers.
The only difference is the col value:

```
vid_0079&pid_1801&col01
vid_0079&pid_1801&col02
vid_0079&pid_1801&col03
vid_0079&pid_1801&col04
```

mirh, 2016-08-23:
#380

closed 2018-01-26 by nefarius