nefarius/ScpToolkit #407: How Can I Delete This Properly

Markeron, 2016-08-09:
After using couple of days, suddenly it stopped working and I haven't been able to make it work no matter how hard I tried. I've decided to go back to scarlet. however I can't access my dongle now, it's light dead. How can I get rid off this completely.

nefarius, 2016-08-10:
[Ta-Da!](../wiki/Manual-driver-removal)

closed 2016-08-10 by nefarius

Markeron, 2016-08-13:
Ta-Da!? You think I haven't checked that. I've uninstall it, but it didn't help me to go back to original bluetooth driver. Do you want me to reinstall this to only manually uninstall it, because if it works I will.

Jan200101, 2016-08-16:
@Markeron find your Bluetooth device in your Device list and uninstall / reinstall its drivers. That should fix the problem.

Markeron, 2016-08-20:
It won't reinstall the drivers, because it can't find the drivers. Something is preventing form it, it's somehow still slave to this piece of shit even though I deleted everything about it both auto and manually. Goodbye cordless gaming.

digital-pollution, 2017-07-15:
Its not a big deal to uninstall the service, drivers and toolkit, but seriously SCP team no teveryone has technical knowledge and to be fair your software isn't very polished, maybe don't provide an installer without the uninstaller.