nefarius/ScpToolkit #408: Windows 10 Anniversary Update - driver not SCP driver signature not recognized

Keefaetana, 2016-08-11:
I recently made a fresh install of Windows 10 with the Anniversary Update included and then attempted to install ScpToolkit v1.7.277.16103-BETA. At first it wouldn't even install properly because Microsoft changed the rules on signing drivers (http://www.thewindowsclub.com/driver-signing-changes-windows-10). I then disabled the driver signature verification, after which installing and using the driver worked perfectly. Now, after rebooting, the driver no longer works. I checked Device Manager and this is what it says:

"Windows cannot verify the digital signature for the drivers required for this device. A recent hardware or software change might have installed a file that is signed incorrectly or damaged, or that might be malicious software from an unknown source. (Code 52)"

Is there any way to permanently disable the driver signature verification?

Thanks a lot!

charlesmilette, 2016-08-11:
As mentionned in #266, you have two choices:

- Disable secure boot
- Run `bcdedit /set loadoptions DDISABLE_INTEGRITY_CHECKS` in a admin command prompt. Please note that DDISABLE is not a typo.

Keefaetana, 2016-08-11:
Ah, I hadn't noticed that. Thanks for the help, charlesmilette!

closed 2016-08-11 by Keefaetana

rougetimelord, 2016-08-11:
That's a bandaid fix, many users want to keep secure boot on (me)

nefarius, 2016-08-12:
@rougetimelord I'm in the middle of the process of getting a compatible certificate.

rougetimelord, 2016-08-12:
@nefarius 👍 I just don't want to fiddle with it and don't have issues (yet) 😜

TheWingedCherryPie, 2016-08-28:
I tried the bcdedit command, nothing changed.