nefarius/ScpToolkit #412: Missed support for Window Server/XP x64

fedor4ever, 2016-08-18:
Why missed support for Window Server/XP x64?

mbc07, 2016-08-19:
...because it's a very old and unsupported OS that not even Microsoft cares anymore? We're in 2016, you should upgrade at least to Windows 7.

fedor4ever, 2016-08-21:
Is there a problem with build for XP? I like it and win10 also good
Visual Studio 15 can build apps for them.

trojanc, 2016-08-21:
OMG why?! haha. I don't blame or expect any developer to even try building stuff for XP. It's ancient history; forget it existed and continue life...

charlesmilette, 2016-08-21:
Is there still people using Windows 3.1? No

Why should it not be the same for Windows XP, an abandonned OS?

BTW, the way driver works has been redesigned in Vista. It might not be possible to port this on Windows XP.

mirh, 2016-08-23:
> Is there still people using Windows 3.1? No

Yes. 

Anyway, back to the problem.. I don't think it's really something with the driver itself being incompatible ViGEm -newest nefarious driver- for example is based on KMDF 1.9, supported under XP. And original SCP driver, same should work under XP. 

My guess is that scptoolkit is now signed with a SHA-2 certificate, which folks claim to create problems with anything before Vista SP2. 
So.. If I were you, I'd try to download WDK 7.1 (last to support XP), and compile my own driver with blackjack and hookers. 
Thankfully, XP (even 64 bit version) shouldn't require any signing, so it should be ready to be tested.

fedor4ever, 2016-08-24:
Wdk comes with with vs2015. I compile Hello word and it works :-D

mirh, 2016-08-24:
I know there's wdk built-in vs2015, but I'm saying you can't build anything for XP (and perhaps even Vista!) with that.

charlesmilette, 2016-08-25:
How can you compile a Hello World driver with WDK?

fedor4ever, 2016-08-26:
@mirh , how build xp drivers - https://msdn.microsoft.com/en-us/windows/hardware/drivers/develop/building-drivers-for-different-versions-of-windows

fedor4ever, 2016-08-26:
@charlesmilette wery intrigue

mirh, 2016-08-26:
Yes, as I said you'll need WDK 7.1.0 to be installed separately, even with vs2015.

nefarius, 2016-08-28:
That's not directly a driver issue, more of a .NET availability issue; .NET 4.5 doesn't install on XP so a large portion of the toolkit won't run. As for my other driver projects on the other hand... :wink:

fedor4ever, 2016-08-29:
Well, Motion Joy play well for me. No althernatives. May I close this issue?

mirh, 2016-08-29:
*Don't name it!*
But.. Anyway, seeing it's just an installer incompatibility I guess so?

Albeit a bigger *funnier* question arises now: how's @nefarius going to treat all those **248** bugs all probably being fixed in newer build? 😄

charlesmilette, 2016-08-29:
```c#
for (int i = 0, i > 248, i++) {
    closeissue(i);
}
```

mirh, 2016-08-29:
Mhh, I dunno, that seems kind of harsh. And for as much as.. I don't think SCP will be updated anymore, it still would feel kind to replace that closeissue function with pushmessage

Where message is something like: hi, this driver has been surpassed by this - link
Is that enough to solve the problem? If so, feel free to close this issue. 

Then after a week we may proceed to manually close shit.

charlesmilette, 2016-08-29:
Then why not push a message, and set a timer for in a week that will close the issue if not already done or notify @nefarius if there has been a new comment, but the issue wasn't closed?

nefarius, 2016-08-29:
Don't worry; I might not be that active here currently but I won't just run a script to close 'em all :laughing: 
What I learned from this project; don't publish beta releases, it's only stressful for both parties...

charlesmilette, 2016-08-29:
You just destroyed my hope of seeing a beta FireShock driver.

nefarius, 2016-08-29:
@charlesmilette thing is; I will lose my sanity and possibly freak out if I get more "I can't install because it's not signed blah blah..." BS so I try to get things right on the first release. M$ lately also being a b$tch with their new driver signing policy... Some very special people might get an early build though :wink:

closed 2016-08-30 by fedor4ever