nefarius/ScpToolkit #413: ScpToolkit won't work in new computer. Windows 10

etianhelvoid, 2016-08-19:
Hi,

I have a genuine Dualshock 3 PS3 controller (it came with the console) and I seem to be having a bit of trouble, I've got a new computer and I've installed the drivers (and all the dependencies beforehand) just as I did in my old one. Even though I went through essentially the same procedure it doesn't seem to work on the new one.
The controller is detected by the driver installer and the computer. But I can't do anything and I have the settings to light the Pad ID and they just eventually turn off after flashing for a bit.
I've attached the debug files.
[ScpToolkit_Log-Package_Utilizador_1471648426.zip](https://github.com/nefarius/ScpToolkit/files/428045/ScpToolkit_Log-Package_Utilizador_1471648426.zip)

Thanks and good work!

charlesmilette, 2016-08-21:
Issue #51 may be the issue

mirh, 2016-08-23:
Or [this](https://github.com/nefarius/ScpToolkit/issues/224#issuecomment-192291387)

closed 2018-01-26 by nefarius