nefarius/ScpToolkit #416: DS3 goes UP randomly while wired.

Z3phyr, 2016-08-23:
Hi! As the title says, my controller, **ONLY** when wired, tends to go UP at random times. Is really annoying.

I've tried different cables and even another controller (mine could be faulty) but the problem persists. I'm on Windows 10 x64 Anniversary Update. This happens in both the beta release and the stable release.

While using bluetooth the controller works perfectly.

I've made a little video that shows the issue: https://gfycat.com/BossyCourteousEmeraldtreeskink

ShayBox, 2016-08-28:
Seems like a controller issue, may be sending two pins or something in the controller sending a wrong signal, but it may be a software issue, have you tried it on any other pc's or softwares (ds3/4, motionjoy)?

nefarius, 2016-08-29:
Smells like a hardware problem; the USB port on your controller might have become loose and randomly causes issues on the PCB. The software just translates so randomness most of the time emerges from the controller itself.

closed 2016-08-29 by nefarius

Z3phyr, 2016-09-01:
As I said, it happens with two different controllers. (tried another cable and another usb port too)