nefarius/ScpToolkit #419: SCP suddenly stopped working for PCSX2 (pressure sensitivity)

Kuhn-Midori, 2016-08-27:
I use the SCP settings manager thing to enable SCP for PCSX2 but recently, I can't get it to work properly. With SCP enabled in PCSX2, I go to the plugins and try to bind the buttons, but it doesn't even show that the controller isn't even attached. In the general tab in the PAD settings it suddenly grayed out and unchecked the option under game device APIs "DualShock 3 native mode (Requires SCP XInput DLL)." I uninstalled and reinstalled PCSX2 and it still does it. I even uninstalled and reinstalled all of the SCP programs/drivers. Nothing is working. Am I missing something?

[For the record, XInputs 0, 1, 2, & 3 work, but I am aiming for pressure sensitivity from the SCPToolkit  Settings Manager.]

nefarius, 2016-08-29:
What version of SCP and PCSX2 are you running?

Kuhn-Midori, 2016-08-30:
PCSX2 1.4.0 which I believe is the newest stable version, and I use the auto-updater for SCP so now it is 1.6.238.16010. The issue has been harassing me for about a week or two though, and still is. 

If it helps, I tried trying it again, and it seems whenever I exit out of the PAD settings when SCP uses its PCSX2 enabling thing, PCSX2 does its double crash thing.

Kuhn-Midori, 2016-09-19:
Okay, it has been nearly a month, and I'm getting a tad frustrated with all this, so, I'm wondering if I can just get an older version of SCP somewhere and try that.

nefarius, 2016-09-24:
AFAIK PCSX2 1.4.X broke compatibility, I fixed it in my new project.

Kuhn-Midori, 2016-09-24:
which project? Fireshock? 
I tried getting that, but I kept getting errors when I try to install it.

nefarius, 2016-09-24:
Yes, FireShock. I don't know what you tried to install since I haven't released it yet :laughing:

Kuhn-Midori, 2016-09-24:
i downloaded the source ... should've guess that wouldn't of worked for shit ... but at least your working on it. still ... I wonder what happened with scp ... it was working perfect until one random day.

nefarius, 2016-09-24:
Well, builing and installing a kernel driver is a whole different story :smile: I'm only one person working on three coding projects in my free time, there will be updates but they will come at my own pace.

Kuhn-Midori, 2016-09-24:
Fair enough.

closed 2016-09-24 by Kuhn-Midori

mashakos, 2017-02-03:
Will fireshock eventually support bluetooth?

xperia64, 2017-08-08:
I know this has been closed for a while, but I recently experienced the same symptoms, and found a fix for my specific issue so I thought I'd share.

After digging around for a few hours, I noticed some Nvidia streaming-related controller DLLs being loaded before Xinput1_3.dll. I uninstalled Geforce Experience 3.8.1.89 beta, and the driver started working again in PCSX2.
It would appear that at least the newest Beta GFE forces applications to always use Xinput1_4.dll and Xinput9_1_0.dll rather than what they request. If anyone else has this issue and has GeForce experience, try uninstalling it.

Kuhn-Midori, 2017-08-08:
oh my god ... thank you @xperia64 . 
it actually worked, darn you nvidia. now I can threaten people in mgs3 again >:D

joman66, 2017-12-24:
I can confirm the problem still exists up until now. I've been looking for a solution as to why PCSX2 didn't recognize my controller months ago and from what @xperia64 has said, that's still the case. I've deleted my copy of Nvidia Geforce Experience and it now works! Thanks goodness! I don't think Nvidia has any idea what they've done. Hopefully they'll fix it soon. 😄