nefarius/ScpToolkit #425: Fake DualShock 3 found. Trying Workarounds..

gobit-ua, 2016-08-30:
I have 2 joystick (http://www.gearbest.com/game-controllers/pp_279646.html)
[image]

By USB - all ok!
But by Bluetooth worked only black joystick.

In notifications i see next, by USB
[image]
and by BT
[image]

Thanks!

gobit-ua, 2016-09-11:
Helped decision https://github.com/nefarius/ScpToolkit/issues/241
Substitution ScpControll.dll http://forums.pcsx2.net/attachment.php?aid=59733 in the release version
https://github.com/nefarius/ScpToolkit/releases/download/v1.6.238.16010/ScpToolkit_Setup.exe

Thanks for j0shg

closed 2016-09-11 by gobit-ua

lovetheories, 2017-01-27:
Did you change your Vid? Because when I install mine fake drivers, it doesn't change...

gobit-ua, 2017-01-27:
No, I didn't. Only changed dll.

lucastcp1, 2017-04-27:
@gobit-ua 
May the gods bless you bro.
After 1 weak with same problem your info worked to me.
I instaled ver 1.6.238 with the dll substitution and now is working. Thank you very much.

Aditional info,  my setup is:
- Windows 10
- fake ps3 controler. device id: vid_045e&pid_028e
- Bluecore csr8510 a10 dongle