nefarius/ScpToolkit #430: New Broadcom Bluetooth 2016 won't work with first gen PS3 controllers

RogueLizard, 2016-09-03:
I have an official PS3 controller and a brand new Lenovo Thinkpad E565 with a Broadcom BCM2045A0 BT 4.0 radio. Works fine over USB but when I try SCP and MiJ neither can see the BT radio, even when editing the MiJ configuration files. Tried with 3 different controllers - none of them worked. All ov the batteries in each of the controllers have been replaced and sync with the PS3.

nefarius, 2016-09-04:
Don't use MiJ together with ScpToolkit, all hell may break loose and f*ck things up.

RogueLizard, 2016-09-04:
I think you misinterpreted what I stated. MiJ works better than SCPToolkit
at this point. The Bluetooth doesn't work with either MiJ OR SCPToolkit.
There's an or statement there. Installed MiJ, works. System restore back to
before MiJ. Install SCPToolkit. Bluetooth still doesn't work.

closed 2018-01-26 by nefarius