nefarius/ScpToolkit #434: Driver Manager can't load d3d9.dll

Tristan971, 2016-09-07:
The driver install app simply crashes on launch ("... has stopped working, windows is looking for a solution bla bla bla"). Looking at the event viewer of the system d3d9.dll seems to be the breaking point.

As it so happens you just have to copy it from System32 to the installation path of scptoolkit.

Don't know if the last updates of windows 10 make loading some libs different but might need to be looked into.

Here is crahs log from msft event viewer :

Log Name:      Application
Source:        Application Error
Date:          07/09/2016 14:33:37
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      SurfaceBook-Tristan
Description:
Faulting application name: ScpDriverInstaller.exe, version: 1.6.238.16010, time stamp: 0x569253fb
Faulting module name: d3d9.dll, version: 10.0.14915.1000, time stamp: 0x57c12752
Exception code: 0xc0000005
Fault offset: 0x000000000000da26
Faulting process ID: 0x182c
Faulting application start time: 0x01d209040d69e7f0
Faulting application path: C:\Program Files\Nefarius Software Solutions\ScpToolkit\ScpDriverInstaller.exe
Faulting module path: C:\WINDOWS\SYSTEM32\d3d9.dll
Report ID: 067b9264-8073-46e3-8161-c8eee47c8ace
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2016-09-07T12:33:37.043306000Z" />
    <EventRecordID>5196</EventRecordID>
    <Channel>Application</Channel>
    <Computer>SurfaceBook-Tristan</Computer>
    <Security />
  </System>
  <EventData>
    <Data>ScpDriverInstaller.exe</Data>
    <Data>1.6.238.16010</Data>
    <Data>569253fb</Data>
    <Data>d3d9.dll</Data>
    <Data>10.0.14915.1000</Data>
    <Data>57c12752</Data>
    <Data>c0000005</Data>
    <Data>000000000000da26</Data>
    <Data>182c</Data>
    <Data>01d209040d69e7f0</Data>
    <Data>C:\Program Files\Nefarius Software Solutions\ScpToolkit\ScpDriverInstaller.exe</Data>
    <Data>C:\WINDOWS\SYSTEM32\d3d9.dll</Data>
    <Data>067b9264-8073-46e3-8161-c8eee47c8ace</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>

nefarius, 2016-09-24:
That's why DirectX is listed in the README as a dependency...

closed 2016-09-24 by nefarius

Tristan971, 2016-09-24:
I have precised that it's installed. You actually have to install it AND copy the dll into the install folder. And that is the actual "issue" although it's minor.

nefarius, 2016-09-24:
Hang on, if you HAVE to copy the DLL then your DX installation might be damaged; all the DLL should be loaded from the system directory automatically.

Tristan971, 2016-09-24:
Yes I am well aware. That said I had SCP working, then windows updated itself and then it didn't work anymore. So I don't know whether it's windows (very possible) that is retarded or some api that just got deprecated.

nefarius, 2016-09-24:
Odd, Microsofts new update policy sure causes some weird stuff.

Tristan971, 2016-09-24:
Well if you Google up apps not working after anniversary update it seems they changed a whole lot of things. Whether it was intentional or not.

charlesmilette, 2016-09-24:
Try installing DirectX from here: https://www.microsoft.com/en-us/download/details.aspx?id=8109

It's a self-extractible zip. Open the folder where you extracted and run DXSETUP.EXE

Tristan971, 2016-09-25:
I tried that but it tells me that there's no update to proceed with as "an equivalent or newer version of DirectX is already installed".
That said even if it fixed it I wouldn't be able to reproduce the bug.