nefarius/ScpToolkit #441: 2 Bluetooth Dongle

LaBoss, 2016-09-11:
Good afternoon,

I'm here with a little problem , I have two bluetooth dongle equal to this (http://www.dx.com/p/ultra-mini-bluetooth-3-0-usb-dongle-65631#.V9UutJgrI2w ) one for DS3 and other for headsets. When I install the DS3 only install with one pen on but after turning the other can not have access to bluetooth devices, it seems that is with the DS3 driver instead of the normal driver.

It is possible to have both linked pens ? One for the DS3 and another for normal bluetooh call.

Thank you

deadair3210, 2016-09-11:
From what I understand, you are having a issue where you have more than one of a specific Bluetooth adapter and the driver is taking over both of the adapters instead of just one,correct?

LaBoss, 2016-09-11:
Yes it .... Sorry for my english

deadair3210, 2016-09-11:
As far as I know, there is no way to keep Windows from using a driver for multiple pieces of the same hardware. The only WA to use Bluetooth for both normal things and the controller without removing the driver is to get another Bluetooth adapter that's not the same as the one you already have

charlesmilette, 2016-09-11:
I'm not sure, but you can go into the properties of one of the adaptaters, then "update driver" and choose the original driver.

theblackwidower, 2016-09-23:
I guess you'll need an adapter with a different hardware ID to use a different driver for it. this means buying a different model or going with a different manufacturer. I'll admit, I'm not an expert, but I presume Windows links the drivers to the hardware ID. It's the only thing they could reliably link it to, because you easily plug the device into a different USB port. So the only solution is buying a different device.

Might help to do what I did once. Buy a cheap one on eBay. Might be able to get one for a dollar from Hong Kong. Should work.

Atolm4, 2016-09-27:
Hi guys, I have a small amount of expirence with this so I'll add what I know;
I have two bluetooth radios and though purchased as different devices they both identify as CSR8510 A10. It is definitely possible through careful installation of the drivers to get both to work. in my case I had to do the following;
1 - wipe the slate clean. remove all traces of scp and its drivers. Remove the software for your other adapter (if it is 'CSR' or 'Cambridge silicon radio' branded 'harmony' software you may need to remove some registry entries. I will try to find the website that describes exactly which). Restart with no radios plugged in.
2 - install the software and drivers for your audio bluetooth radio, plug it in when required. Restart.
3 - plug in the SCP radio and begin to install SCP. when asked, Try to choose the correct radio. Restart. (if you accidentally install the scp driver to the wrong radio, use windows device manager to 'roll back driver', and reinstall scp on the correct radio before restarting. This worked for me).

If you're lucky, sometimes you will get error messages but everything works. sometimes nothing works. Sometimes its fine for a week and then breaks. sometimes audio is fine but you cant send files from a phone. I often receive the message "warning couldnt get connection handle" when doing anything bluetooth outside of SCP. I'm very interested in getting this working and willing to run tests if there's anything I can do to help. Just drop me a PM or something.

amcsi, 2017-10-07:
I was easily able to solve this by going to `Control Panel\All Control Panel Items\Devices and Printers`
1. Right clicking the bluetooth adapter whose driver you want to change
1. Hardware tab
1. Properties button
1. Change settings button with shield
1. Driver tab
1. Update Driver button
1. "Browse my computer for driver software" (don't worry, you won't actually have to)
1. "Let me pick from a list of available drivers on my computer"
1. Generic Bluetooth Adapter
1. etc.

closed 2018-01-26 by nefarius