nefarius/ScpToolkit #443: High CPU usage introduces severe input lag

lgfrbcsgo, 2016-09-13:
When the CPU usage approaches the 90% mark in CPU-demanding games (like in AC: Syndicate), the input starts lagging serverly. Setting the ScpService.exe process to a higher than "normal" priority in task manager fixed the input lag. Is it possible to set the process priority at start up? I am using Windows 10 and never had problems like this under Windows 7 (even in AC: Syndicate).

k0enf0rNL, 2016-09-15:
@lgfrbcsgo No its not possible to do this at startup unfortunately. You will have to do this everytime you start scp.

lgfrbcsgo, 2016-09-16:
This might be helpful https://msdn.microsoft.com/de-de/library/windows/desktop/ms686219(v=vs.85).aspx

closed 2018-01-26 by nefarius