nefarius/ScpToolkit #448: Bluetooth DS4 auto disconnecting?

Shinjed, 2016-09-17:
https://gyazo.com/bbf3fbe052b958110529e277b7a6f198

This shows what happends in the SCP Server when I unplug my DS4 and attempt to use it through bluetooth.

Tried everything I can think of, several different versions and it all comes out the same, my bluetooth adapter is listed under the supported ones, it's the Asus BT4.0 one in the list.

Hoborific, 2016-09-26:
I have this issue with identical log, it was working for a few days but now even with a cleanwipe/full reinstall i can't get it to connect via bluetooth.

weakfl, 2016-10-29:
I had the same problem, here's how I solved it:

1. Open `ScpSettings.exe`
2. Go to `DualShock 4 specific`
3. Check `Always re-pair controller on Usb connection`
4. Close `ScpSettings.exe`
5. Connect controller(s) via USB
6. Disconnect USB and connect via Bluetooth

It should work now and you can open `ScpSettings.exe` again and disable `Always re-pair controller on Usb connection`

closed 2018-01-26 by nefarius