nefarius/ScpToolkit #449: Just doesn't work on my system.

IvanB303, 2016-09-19:
Hi there

I have Windows 8.1 and an original DualShock 3 controller that came with my PS3.

I followed the steps mentioned in the installation page. 
Plug in controller before doing anything. Windows 8.1 detects and installs it's own drivers
it comes up as Playstation Dual Shock 3 which is what it is.

Running ScpToolkit Installer as admin it completes fine.
I open the Install driver and this is when I tried a few things.
It detects a DS3, an integrated webcam and a Bluetooth controller.

Even just installing the controller and using it connected by the wire doesn't work.
Nothing happens. SCP Monitor detects it as being plugged in and charging but that's as
far as it goes I opened the Analyser and nothing happens.

Cheers

charlesmilette, 2016-09-19:
Do you happen to use an EFI system?

It might be related to #266

IvanB303, 2016-09-19:
Hi

What's a EFI system ? The thread you posted mentions it all works fine on
Win 8.1 that the issue is for Win 10. I have Win 8.1 and it doesn't work
even when I'm trying to just to use it plugged in. What info do you need ?
Will try again with screenshots and post it. Thanks

Ivan

IvanB303, 2016-09-19:
Hi

It never asks me about driver signatures or anything when I do the
installation. It seems to all install successfully but the controller
doesn't work, I see the four blinking lights like it's charging or looking
for a host but that's it.

Regards
Ivan

charlesmilette, 2016-09-19:
Okay.

It doesn't seem to be a driver signature problem then.

Try to run the standalone server

IvanB303, 2016-09-21:
Hi Charles

I seem to have gotten it sort of working. All I did was install the
Bluetooth driver and DS3 driver the problem is that the installer app gets
frozen and doesn't complete the operation. The controller seems to work
when I test it using the SCP tools tester. Just one light is on the DS3
controller. I'm using it wired ie not via Bluetooth.

Regards
Ivan

charlesmilette, 2016-09-21:
The 1 light is normal, because the controller is player 1.

Can you try by Bluetooth now?

IvanB303, 2016-09-21:
After I reboot computer I have to wipe installation and re do it but wired
method seems to work will try wirelesslly

Ivan

closed 2018-01-26 by nefarius