nefarius/ScpToolkit #450: After driver install touchpad is completely gone

JohnP80, 2016-09-20:
I was using the driver utility and when I hit install, the program did its thing and then it never finished. It said it would take up to five minutes, so I let it sit for ten or so. When I closed the program my touchpad just didn't work anymore. When I mean "didn't work" I mean that my mouse disappeared off of the screen and it just didn't exist anymore in the device manager. A quick response would be great, as I use this computer for college.

charlesmilette, 2016-09-20:
Have you checked every box when choosing the device to install?

Sonickyle27, 2016-09-21:
Make sure when installing the drivers that you only select your controllers and your external bluetooth dongle. Selecting anything else will cause the device to no longer work. Don't tick everything.

nefarius, 2016-09-24:
*Sigh*

closed 2018-01-26 by nefarius