nefarius/ScpToolkit #451: ScpToolkit disabled mouse and keyboard

Kryllic, 2016-09-21:
So I tried installing this program to use my Dualshock 4 controller with certain games (I'm on Windows 10, btw), but while installing I noticed an error or 2 pop up, and then when it finished, my mouse and keyboard stopped responding. I restarted my computer to see if it would fix it, but when I got to my login screen, my cursor was nowhere to be seen, and there was still no response from it or my keyboard. I've tried entering safe mode (which my mouse and keyboard work in the bios advanced settings), but the same issue occurred when my login screen popped up during safe mode. I have no idea what to do, I virtually can't use my mouse or keyboard at this point.

charlesmilette, 2016-09-21:
If you checked all the boxes in the installer, __it also overwrites any USB devices currently plugged__.

Get another mouse and keyboard, [clean wipe](https://github.com/nefarius/ScpToolkit/wiki/Manual-driver-removal) and only check the checkbox for your Dualshock 4 controller, under this category:
[image]

If you have a Bluetooth dongle, just check it under this category also:
[image]

You also need this one checked so that it actually works:
[image]

Kryllic, 2016-09-21:
Thank you so much, I was able to delete the drives and gained access back to my K&M.

As for installing the correct drivers, do you mean I select only the "Install Dualshock 4 driver" option, and nothing else under the "Choose Dualshock 4 controllers to install" drop down? Cuz those all look like the USB ports that my other devices are currently using

charlesmilette, 2016-09-21:
No, I mean you select only your DS4 under the dropdown.

That's exactly what is under this dropdown. All USB devices. So checking "Install DualShock 4" driver checks everything under the dropdown, and you know what happens then.

Kryllic, 2016-09-21:
So will it overwrite that USB port so that only the DS4 will work with it?

charlesmilette, 2016-09-21:
No.

You'll be able to use your DS4 on any port that you want, and you can plug something else where the DS4 was.

Windows recognizes the controller as the same no matter in what port you plugged it in. And can recognize any device you connected no matter what you plugged in before.

Kryllic, 2016-09-21:
Ah I see,  so before, I basically made my computer think all my devices I currently had were a ds4. 

Thanks for the info, friend, it truly helped a lot. I'll post anything new if I run into another problem

Kryllic, 2016-09-21:
Alright, I just want to make sure I'm selecting the correct port here, but my list currently shows "USB PnP Audio Device (Interface 0)", "USB Reciever (Interface 0)", "Wireless Controller", "USB Reciever (Interface 1)", "Desktop", "USB Laser Game Mouse (Interface 2)", "USB Laser Game Mouse (Interface 1)", "External", "USB Laser Game Mouse (Interface 0)", "XBOX ACC", and "USB Reciever (Interface 2)". I currently have the DS4 plugged into a USB Hub via micro-USB. Not sure which one to choose, I'm kinda thinking one of the USB Receivers or the XBOX ACC

charlesmilette, 2016-09-21:
It's the one called "Wireless Controller"

Kryllic, 2016-09-21:
Ah, alright, thank you again!

Kryllic, 2016-09-21:
So, the Wireless Controller turned out to be my wireless keyboard

charlesmilette, 2016-09-21:
Well then I don't know what device would be the DS4.

Try running the installer without it plugged in, and compare with when it is plugged in.

The DS4 should then be the one that wasn't here.

Don't forget to clean wipe before.

Rasive, 2016-10-17:
It's still not working. After checking my ps3 controller and bluetooth dongle, it for whatever reason, keeps hijacking my keyboard. Will provide more info if needed.

Rasive, 2016-10-18:
Ok - I found a temp fix, simply install without keyboard and/or mouse (sounds weird i know) and it works afterwards. Probably because it messes up the bus addresses

genkidama20, 2017-01-19:
Sorry to hijack your post, but I had the exact same error, except I'm trying this on my laptop.  So, there was no way for me to disconnect my keyboard and mouse.  I managed to get into device manager via on-screen keyboard with my touchscreen and uninstall (checked the "delete driver" box as well) the extra keyboard driver under the "libusbK USB Devices" section.  I didn't check any boxes but my controller boxes, but scptoolkit installed a keyboard driver anyways.  Weird.

Hope this helps someone.

maur7c7o, 2017-07-31:
Thanks  charlesmilette I have the same problem with my mouse, now everything is ok.

lawiiminae, 2017-10-25:
Halo, I am using an external mouse on my laptop to game, however, now the mouse will not work after installing SCP toolkit. I followed the above instructions: clean wipe, install DS3 pad only from the drop down but my mouse will still not work... 
does this mean i need a new mouse?

NinaRoseXO, 2017-12-14:
Hello I followed this method via the "clean wipe". But I made a mistake I uninstalled drivers from "Universal Serial Bus controllers" instead of "Universal Serial Bus devices". So not only have I screwed my keyboard and mouse with the SCPToolkit, I have tried to resolve the issue with this method, making a mistake which has completely screwed everything and now I can't do a thing? I don't have another keyboard or mouse to try it with... once I get them I just hope they work! All I wanted to do was play with a controller.. wasn't worth all this! :(

lawiiminae, 2017-12-15:
Don't worry men, we all went through such.

MarioMasta64, 2017-12-15:
seems everyone has this error

gameyakuza, 2017-12-30:
Hi, I have a different problem, my DS4 seems to work as a keyboard, buttons seem to type "enter" and other commands, and the right stick controls the cursor. How do I disable it?

michmich59, 2018-01-16:
SCP toolkit is probably one of the worst software that have ever existed. 

Once you'll get back to normal, uninstall every piece of it, and stay far far away of any software that will be coded by the same persons.
Seriously, deleting driver without permissions ? what kind of developer codes such thing

Jesus

closed 2018-01-26 by nefarius