nefarius/ScpToolkit #453: [Suggestion] Give volume mixer entry a name

InstantMuffin, 2016-09-23:
So recently I discovered that in Win7's volume mixer ui I had an entry in italic font named "Name Not Available". I was suspecting malware (like the google results did) but scans were negative. Unfortunately Windows does not have a feature to let you find the processes associated with those entries.
The only way to verify this was to run a snippet found here:
http://stackoverflow.com/questions/20938934/controlling-applications-volume-by-process-id

I'm lucky enough to come up with this, however others are not. I ended up finding threads like these where people suspect malware and they don't find anything, with the logs however showing that they do run the scptoolkit.

It would be really nice to fix this and spare users the trouble of suspecting a compromised system and ending up formatting their drives.

deadair3210, 2016-09-23:
Afaik there is really no way to fix this easily, as the reason that the volume applet shows name not avaliable is that the scp service runs with admin privileges, which means that the sound applet can't pull the name

charlesmilette, 2016-09-23:
As a workaround, you can turn the sounds off in the toolkit's settings to remove the entry from the volume rocker.

InstantMuffin, 2016-09-23:
Yeah I was suspecting something like this.
I posted in one of the results I could find and raised awareness to this oddity and suggested a way how to properly look into these entries.
https://forums.malwarebytes.org/topic/185011-windows-10-name-not-available-audio-mixer-please-help/
My hope is that this goes up in the search results and/or people will start tackling this issue differently on these forums.

Since I see charlesmilette's reply popping up as I'm typing this, maybe a mouseover caption or something like that should be added to the corresponding entry in the toolkit's settings.

nefarius, 2016-09-24:
I'll just remove the audio BS entirely since it seems to just cause headaches. I still stand by my words though; it's a Windows permission issue, not mine :wink:

deadair3210, 2016-09-24:
It is definitely an issue that stems from the permission model in Windows. There is no way for non priviliged code (the volume sliders), and the service where the sounds originate from(SCP Service)

nefarius, 2016-09-24:
A common workaround would be to create a separate process which runs in the current users session that creates the sound player. Not worth it though, I'll kick the entire feature.

closed 2016-09-24 by nefarius