nefarius/ScpToolkit #457: ScP not working with DS3 in Windows 10

FliccC, 2016-09-25:
1. I plug in my DS3 via USB. The 4 led lights are blinking. I hear the notification sound of windows, but the device doesn't show up in the window "Game Controllers". 

2. I install the latest ScP Package with Bluetooth and DS3 drivers checked. Installation completes successfully.

3. ??? Nothing works. The 4 led lights are still blinking and the device shows up nowhere. It also doesn't appear in the win10 bluetooth pairing window

Any help? I struggle to find what the next step is supposed to be as I don't get any error messages or such. The controller works fine on the PS3 btw.

charlesmilette, 2016-09-25:
Have you ran the driver installer?

It's in the start menu, under the folder "ScpToolkit"

FliccC, 2016-09-25:
I did that before, so now I ran the driver installer again with right click "as administrator" and this time it worked!

So, everything fine here. If anyone should encounter a similar issue: **run as administrator and reboot!**

closed 2016-09-25 by FliccC