nefarius/ScpToolkit #458: Not working

rojasmaia, 2016-09-26:
Hi! I've been trying to get this to work. I also can't check the Device's properties because nothing is listed. Any help would be appreciated! Thank you.

[ScpToolkit_Log-Package_ASUS PC_1474852953.zip](https://github.com/nefarius/ScpToolkit/files/492053/ScpToolkit_Log-Package_ASUS.PC_1474852953.zip)
[image]

charlesmilette, 2016-09-26:
After installing the app, there's a driver installer to run. You'll find it under "ScpToolkit" in the start menu

rojasmaia, 2016-09-26:
Yes, I did that the first time I tried to install. I already tried uninstalling and reinstalling. "Force Driver Install" takes ages (took 20 minutes and still nothing)

closed 2018-01-26 by nefarius