nefarius/ScpToolkit #464: Beta Installer doesn't recognize Bluetooth adaptor

secondsabre, 2016-09-30:
I have a generic-brand Bluetooth USB dongle that the new beta installer isn't recognizing.

The current stable installer sees the device and I can install the driver without any issue, everything works fine.  The new installer, though, seems to automatically filter the list of available devices and my Bluetooth adapter isn't listed. There doesn't seem to be a way to manually select or force it, either?

Everything is hunky-dory on the stable version, so I figure it's probably something to do with the new installer's device filter?  I can provide any information about the adapter you need, just lemme know.

charlesmilette, 2016-09-30:
The best solution would be to list all Bluetooth adaptaters installed (like
device manager does under the Bluetooth dropdown) and offer checkboxes to
choose the adaptater(s) to "enslave".

What do you think, @nefarius ?

But on the beta ATM only shows devices that are proved to work correctly
(aka whitelisted)

nefarius, 2016-10-08:
@charlesmilette the last time I introduced checkboxes and free will to the driver installer I ended up with lots of angry comments from idio... I mean users which disabled mice and keyboards 😉 no software can ever replace a good functioning human brain.

Gistix, 2016-10-29:
What was last version that had this feature? I'm having a similar problem...

closed 2018-01-26 by nefarius