nefarius/ScpToolkit #468: Sony Wireless adaptor

Silent1990, 2016-10-03:
Can you do a Driver for DS4 Sony Wireless adaptor?
I plugged in and it uses the default windows driver, when i checked in joy.cpl it shows as DualShock@4 USB Wireless Adaptor test is the aix and buttons work...all is good...but when i opened a game such as Fallout 4, Dying Light or No Man's Sky i don't get any input. The Games is supporting DS4 light bar and button prompts.
I was trying all:
-inputmapper(without x360 emulator and exclusive mode)
-DS4Windows(same as above)
-with  scp driver for bluetooth latest 
And nothing changed then i remove all of the app and driver and still nothing...

Can you do a emulator as the x360? i mean from DS4 wireless adaptor to DS4 wired mode that the Games supporting DS4 light bar and button prompts?
Here are some screenshots
[image]
[DS4 Adaptor.zip](https://github.com/nefarius/ScpToolkit/files/506791/DS4.Adaptor.zip)
[wired DS4.zip](https://github.com/nefarius/ScpToolkit/files/506793/wired.DS4.zip)

nefarius, 2016-10-08:
[You could gift me one](https://www.amazon.de/gp/registry/wishlist/7B1BMHKGWI7X/) and I make a driver 😉

Silent1990, 2016-10-08:
I don`t know how to do that because im from europe/germany.

nefarius, 2016-10-08:
I'm from Europe, Austria 😆

Silent1990, 2016-10-08:
Vom wo von Österreich kommst du denn?

nefarius, 2016-10-08:
Oberösterreich 😄

Silent1990, 2016-10-08:
Linz?

nefarius, 2016-10-08:
Yepp

Silent1990, 2016-10-08:
was brauchst du denn das ein treiber machst?(ich meine info von datei, oder)

nefarius, 2016-10-08:
Ich brauch' entweder die Hardware selbst oder einen PC mit dem Ding angestöpselt auf dem ich meine Analyse-Tools starten kann.

Silent1990, 2016-10-08:
duch den teamviewer ist das möglich?

nefarius, 2016-10-08:
z.B. ja

Silent1990, 2016-10-08:
ist ok windows 8.1 oder besser windows 10?

nefarius, 2016-10-08:
Egal

Silent1990, 2016-10-08:
668159453   Kennwort 7980

nefarius, 2016-10-08:
Bist du Irre, poste sowas doch nicht öffentlich 😥 außerdem hab' ich den 10er lizenziert, 11er geht nicht.

Silent1990, 2016-10-08:
wie soll ich dann senden den kennwort

nefarius, 2016-10-08:
z.B. [hier](https://gitter.im/nefarius/ScpToolkit)

closed 2018-01-26 by nefarius