nefarius/ScpToolkit #469: Can't seem to get the drivers to load

Kyujaku, 2016-10-04:
Currently using up to date version of Windows 10, and I get this in my log:

[ScpToolkit_Log-Package_Colin_1475560903.zip](https://github.com/nefarius/ScpToolkit/files/508741/ScpToolkit_Log-Package_Colin_1475560903.zip)

Anything someone can do to help would be greatly appreciated!
(My knowledge of coding and programming is very limited so I apologize if this is a simple fix)

evildog1, 2016-10-12:
Have you selected your controller in Driver Installer?

[image]

Kyujaku, 2016-10-13:
Yes I did, even tried to Force Driver install but it didn't help either.

closed 2018-01-26 by nefarius