nefarius/ScpToolkit #476: controller plugged but not working

jasonh57, 2016-10-16:
I've been trying to install the driver many time (it always install my keyboard or mouse with my PS3 controller) and when I try the controller it doesn't work even if it's lighted to say that it's well plugged. I can say that I'm a little mad after 3 time that I turn off my computer and I even reinstalled scp toolkit but still it doesn't work.

Ulquios, 2016-10-16:
I am having barely the same issue.
It just installed the controller's drivers in my case.

The controller is well plugged to my PC but SCP don't recognize it. For SCP, there is no controller connected ...

If you have any news on how to make it work please post it here, i'll do the same.

jasonh57, 2016-10-16:
kay we'll just wait for someone ready to help together

closed 2018-01-26 by nefarius