nefarius/ScpToolkit #483: stick moving on its own

awott10, 2016-10-21:
Every so often the right stick will do a short of "jitter" like it is being flicked quickly in small increments for some reason. I've increased the dead zone all the way to no effect. Not sure what else I can try.

awott10, 2016-10-21:
The d pad also seems to get stuck on the left.

This is not constant but every few minutes

wudustan, 2016-11-14:
This may be an issue with software but it's also a common problem with gamepads in general. Search 'ghosting' on google. It will give you ways to test whether you're actually ghosting of whether you're getting a software error. Also, try cleaning the analog sticks for gunk which may be triggering the motion.

closed 2018-01-26 by nefarius