nefarius/ScpToolkit #493: New Dual Shock 4 v2 (2016) not recognized by installer

pantra64, 2016-10-29:
My new Dual Shock 4 v2 (the one that got released together with the PS4 slim in 2016) is not appearing in the uninitialized controllers section of the driver installer. It works instantly though with InputMapper 1.6.10.19991.

pantra64, 2016-10-29:
I just found out that the new Dual Shock 4 has a different ID which isn't present in the ds4controller.inf. So far the inf only contains entries for USB\VID_054C&PID_05C4, but the new one is USB\VID_054C&PID_09CC. So I suppose this would have to be added for the controller to be recognized.

dsrdakota, 2016-12-30:
USB\VID_054C&PID_09CC please add

Genesis-Legion, 2017-01-17:
Any idea if this is something we can fix manually? or would a fix need to be added and recompiled?

ajmi982, 2017-11-26:
this was never fixed huh?

madhesoe, 2017-12-03:
please add the USB\VID_054C&PID_09CC
i want my ds3 and my ds4 working flawlessly.

the latest installer cannot detect the new controller

closed 2018-01-26 by nefarius