nefarius/ScpToolkit #495: Not allowing someone to choose which bluetooth dongle in the latest Beta

prataczak, 2016-10-30:
I had to reformat my computer, and when I was setting everything back up, I must have been tired because I didn't realize my onboard bluetooth was getting hijacked as well as the dongle.

This is an oversight. I wasted hours trying to get the onboard dongle working - reinstalling default motherboard drivers and utilities and whatnot, and ended up just doing a restore which cost me even more hours to get everything set back up.

Earlier versions allowed you to choose, why would you remove that? Why would you want to write to all bluetooth dongles?

prataczak, 2016-10-30:
Just wanted to say also that your software is incredible. Doesn't seem right to gripe without giving you props.

Please, re-add the ability to choose which Bluetooth device is being affected. I should have taken more care in what I was doing but it simply doesn't make sense to affect more than one.

prataczak, 2016-10-30:
If someone could direct me on how to install this without having it affect my onboard bluetooth I would be very grateful. I've tried disabling it in the BIOS and couldn't find anything. I've tried disabling radios, services, I've even tried installing it, and then trying to fix the onboard bluetooth.

It's also worth noting that uninstalling this does not revert the onboard bluetooth back to normal like an older version I've tried. Once I install it once, the only way I've figured out to get it working again is to do a restore before I installed it.

charlesmilette, 2016-10-30:
Try reinstalling the Bluetooth drivers from your manufacturer's
website after installing ScpToolkit. The installer should remove Scp
only from the onboard Bluetooth and leave it on the dongle.

charlesmilette, 2016-10-30:
Unplug the bundle.
Go into the device manager, all Scp devices should be under "Universal Serial Bus Devices"
Find "Bluetooth Host (ScpToolkit)" and right-click on it.
Select "Update drivers"
Select "Search for available drivers on Windows Update"

Should do the job

prataczak, 2016-10-31:
That seems to have worked... but now I'm having another issue, perhaps you can help me.

[image]

I've been restoring images left and right, and now I'm getting this every time I try to test pressure sensitivity. Please help if you can.

charlesmilette, 2016-10-31:
Sorry, but I can't. @nefarius may be able to help you with this one.

prataczak, 2016-10-31:
I figured it out!

I was manually re-patching the PCSX2 folder after installation of the latest beta 3 driver.

Doesn't matter if it's a fresh install of the patch, or overwriting an existing one - if you apply the patch, it will break functionality and you will see pop ups like in my pic. If, however, you get everything working with the stable driver, and then switch to the beta, the beta will work fine, so long as you never apply the patch.

Wow, that took me WAY too long to figure out. What a headache.

But thank you sir for cutting my frustration and wasted time down significantly. I may never have thought to look where you pointed me to. I was looking in Bluetooth. I didn't follow your steps exactly because my setup is slightly different, but you pointed me in the right direction and it was easy from there.

Thanks again!

closed 2018-01-26 by nefarius