nefarius/ScpToolkit #497: I can't pair via Bluetooth

charlesmilette, 2016-10-30:
So I've got a new Intel Bluetooth chip (AC 3160) and installed ScpToolkit (the latest beta) on it.

Altough the controller (Genuine DS3) works via wire, it can't pair to the Bluetooth chip automatically. The pair button is also grayed out on the standalone server:

[image]

charlesmilette, 2016-10-30:
I just checked. It did got paired automatically, but it can't connect via Bluetooth.

Box-Of-Hats, 2017-02-01:
Im also having this same issue

googleygoggles, 2017-06-18:
Similar issue here, Intel AC 8260 WiFi and Bluetooth chip here... Seems to affect this series in particular. I just get Host Address <Error> instead and 4 flashing lights on my genuine PS3 Controller. My laptop's Bluetooth chip:

VID: 8087 PID: 0A2B

I have to constantly uninstall and reinstall the Drivers everytime I need to use it :(

closed 2018-01-26 by nefarius