nefarius/ScpToolkit #499: Pressure sensitive patch not being applied correctly/faulty XInput1_3.dll file

prataczak, 2016-10-31:
I spent hours trying to figure this out. But if you specify a directory to patch with the ScpToolkit Settings Manager (using ScpToolkit v1.7.277.16103-BETA) to patch, pressure sensitivity is non functional, and it will break pressure sensitive functionality if it worked with a previous version.

The culprit is the XInput1_3.dll file. I know this, because I applied the patch with a previous version of the SCPToolkit to an alternate folder where PCSX2 had been installed, and it worked fine. I then moved the XInput1_3.dll file to overwrite the one that the the latest SCPToolkit installed, and everything worked.

Here is a screenshot of the kind of errors I was getting.

[image]

NomadGSF, 2017-03-03:
Mind uploading the working XInput1_3 file you used? I'm running into the same issue and Nefarious seems to have dropped support for this.

prataczak, 2017-03-03:
Rename the file to XInput1_3.dll, had to do it this way to upload it.

[XInput1_3 - Copy.txt](https://github.com/nefarius/ScpToolkit/files/815723/XInput1_3.-.Copy.txt)

closed 2018-01-26 by nefarius