nefarius/ScpToolkit #500: BT 000000 none

Abbieben, 2016-11-02:
I keep getting BT 000000 none error when i try to pair my gamepad with my PC, but it works well in USB mode, please can you release an update to correct this issue

charlesmilette, 2016-11-02:
Not giving enough information and asking for a fix will lead you
nowhere. What's your Bluetooth dongle? Can we have a debug package?

Abbieben, 2016-11-02:
[ScpToolkit_Log-Package_Abbie_1478077550.zip](https://github.com/nefarius/ScpToolkit/files/566094/ScpToolkit_Log-Package_Abbie_1478077550.zip)
The attahed is a copy of my SCP debug log, 
I read somehwere that  a "modded SCPControl.dll" will allow Bluetooth communication with the fake DS3 pad, i tried it using it with the SCPdriver i got from here "http://emulation-general.wikia.com/wiki/SCP_Driver_Package" and it worked, but when i tried the same method with the SCPtoolkit, the SCPtoolkit driver instaler did not run, i guess due to the different version of the scpcontroll.dll. 
i hope you find this info useful for my problem

Abbieben, 2016-11-02:
@charlesmilette i was hoping you will reply me

charlesmilette, 2016-11-02:
Sorry.

Try installing the driver, then put the modded dll instead.

Abbieben, 2016-11-02:
@charlesmilette it tells me to consult log for errors, with still no luck
[ScpDsxInstaller_Abbie.log.zip](https://github.com/nefarius/ScpToolkit/files/567078/ScpDsxInstaller_Abbie.log.zip)

is this project going to come with an update in the near future?

charlesmilette, 2016-11-02:
After inpecting the log, I determined you have a Broadcom BCM2070 Bluetooth chip, which is totally compatible with Scp. The problem is, by logic, the modded DLL or the way Bluetooth works with your fake controller. Contacting the modder will provide you better help.

closed 2018-01-26 by nefarius