nefarius/ScpToolkit #506: Change Host Address

TechnoDaft, 2016-11-10:
So I got 2 Bluetooth CSR dongles {for using mouth and keyboard on ps3 originally} but I got 1 hooked up to my PC and 1 hooked up to my kids pc and 2 ps3 controller but SCP shows the same host address on both pc's and most of the time both controllers want to connect to the same pc and i'm wondering if there is a way to change the host address so there isn't 2 difrent hosts the controllers can connect to with the same address....thanks
[image]

charlesmilette, 2016-11-10:
Host address is tied to hardware. Generally, it is impossible to change it.
The best way to change it is to get a new dongle.

TechnoDaft, 2016-11-10:
That's what I thought kinda retarded the dongles have the same mac -_- thanks tho

TechnoDaft, 2016-11-10:
I bought the CSR8510 since it was compatible with Gimx and they said had the ability to change the mac address {which is why i guess they all had the same mac address} and figured out how to change it.
Had to uninstall the SCPtoolkit drivers then was able to use this tool **https://macaddresschanger.com** to change the mac address then reinstalled the drivers and everything is working fine now. Thank alot tho
[image]

closed 2016-11-10 by TechnoDaft