nefarius/ScpToolkit #507: Please add Bluetooth Hardware ID for detection

slippils, 2016-11-11:
Please add:

USB\VID_0A5C&PID_217F

to the bluetooth hardware list.  I was able to make it work in the newest release by manually adding it to the BluetoothHost.inf file and enabling driver testing mode in Win10, then manually updating the driver on it using the BluetoothHost.inf file.

This device was labled as a Thinkpad Bluetooth 3.0.  I've also seen the hardware id associated with Broadcom 43142.

charlesmilette, 2016-11-11:
I've had some issues with a Broadcom 43142 and Scp. Does bluetooth still works after a restart?

slippils, 2016-11-11:
Yes.  I just tested it.  Full shutdown and restart, bluetooth still works.  Tested my DS3 in retroarch with super mario world just to make sure it works correctly, everything seems fine.  I was previously using it in Ubuntu with qtsixa without any issues as well.

closed 2018-01-26 by nefarius