nefarius/ScpToolkit #513: NEED ANSWER QUICK!!! Bluetooth Broken

MrBreadWater, 2016-11-19:
Hello, I recently installed the SCPToolkit, but now, my Bluetooth menu has disappeared. It was after I restarted it. I even scanned for devices in Device Manager, but no cigar. I need help, quick! I use the Bluetooth for my headphones, my fitbit, and my arduino. I need it back SOON! What can I do???

charlesmilette, 2016-11-19:
Did you installed Bluetooth on SCP also?

It's expected behavior.

[Do a clean wipe](https://github.com/nefarius/ScpToolkit/wiki/Manual-driver-removal) and install it without the Bluetooth drivers.

charlesmilette, 2016-11-19:
Just press on the close button under the comment box to close it.

MrBreadWater, 2016-11-19:
There is no close button.

charlesmilette, 2016-11-19:
[image]

MrBreadWater, 2016-11-19:
Oh.

closed 2016-11-19 by MrBreadWater