nefarius/ScpToolkit #516: DS4 stops working when wired

Gallion, 2016-11-20:
SCP monitor says that it is connected. However it also says : Pad One : DS4 40:1B:5F:9D:D5:A4 - Usb 00000000 None
The controls do not respond when I test the Controller settings in Windows.
Doing a clear wipe and reinstalling Scp makes it work again wired. But when I reconnect it after I've used up its battery, it doesn't work.

cylgalad, 2016-11-21:
Did you try to unplug/replug it?
When I need to charge it I've noticed that it stops working after being plugged but gets working again when I unplug/replug it.

Gallion, 2016-11-22:
I have tried but it didn't work either.

Gallion, 2016-11-28:
The controller stopped working completely. Scp could detect it, but the light on the controller was on (I set it to 0% brightness in scp) and controls didn't work. I had to fully uninstall scp and install it again to get it to work.

joaofar, 2017-01-29:
I have the exact same problem, it only works wired when the service starts, on a system reboot for example. If I then disconnect the usb cable and connect it again, it is impossible to get working via USB. The popups do say that a ds4 controller is plugged via USB, so it is recognized but it just doesn't work. Bluetooth works as it should though

Kazekai, 2017-04-08:
I have the same problem and have had it for MONTHS. I'm on windows 7. Sometimes uninstalling the controller driver works, sometimes restarting the SCP service works, but a lot of the time, both these processes just hang and create half-alive versions of themselves I can do nothing about and a reboot is required. I am SICK and TIRED of rebooting my PC EVERY FRIGGIN TIME my driver breaks. This is apparently not an isolated issue and the OP has been up since november, why hasn't this been replied to by a dev yet?

hiddenhokage, 2017-06-23:
Having the same problem. I created a desktop shortcut .bat that restarts the service, but this is really a huge pain especially if I unplug my controller by accident while playing a game.

Here's the script I used:
`net stop "DS3Service"
net start "DS3Service"`

closed 2018-01-26 by nefarius