nefarius/ScpToolkit #517: Windows 10 Anniversary Update purges drivers and service on restart.

BioGenx2b, 2016-11-21:
Every time I restart Windows 10 Anniversary Edition, I have to plug the controllers back in and reinstall everything. The Windows service is marked for deletion and can't be started, and nothing works until you run through it once. Afterwards, everything is fine until the next restart.

This wasn't an issue with previous versions of Windows 10.

DanHolli123, 2016-12-02:
this is a Windows problem nothing with this program can fix that, except for maybe adding a special program that auto starts and reinstalls the service but the drivers are handled by windows

charlesmilette, 2016-12-02:
I've never had this issue with the Windows 10 Anniversary Update.

DanHolli123, 2016-12-02:
but many computers have problems when updated to Win 10 Anniversary Update,
I work at a tech place

BioGenx2b, 2016-12-02:
If there was a command line argument I could invoke to automatically
reinstall the services and drivers, it'd be enough. I constantly have to
reinstall it by hand on my Steam box and another PC at a friend's place.

*BioGenx2b*
*Twitch Broadcaster | Tech Aficionado*

*Bad News Gaming Stream TeamTEST Gaming Stream Team*
*AMD Red Team Plus*
*http://twitch.tv/BioGenx2b <http://twitch.tv/BioGenx2b>*

charlesmilette, 2016-12-02:
The service can reinstall the drivers, but if the service is deleted, it
can't.

BioGenx2b, 2016-12-02:
> BeEBsm1EKkhYPDks5rEFXJgaJpZM4K4SDf

That's what I'm talking about. I go to restart the DSx service and it says
that it's flagged for deletion. I have to reinstall everything. If I could
automate that with a script, I could at least just set that to autorun and
resolve the issue. I don't mind the extra 15 seconds it takes to do that.

I don't know why it does this on a fresh install though.

*BioGenx2b*
*Twitch Broadcaster | Tech Aficionado*

*Bad News Gaming Stream TeamTEST Gaming Stream Team*
*AMD Red Team Plus*
*http://twitch.tv/BioGenx2b <http://twitch.tv/BioGenx2b>*

nefarius, 2016-12-03:
@BioGenx2b would you mind removing this signature/footer/ad thingy that attaches to your posts? It's not helping anyone, thanks.

BioGenx2b, 2016-12-03:
I'm replying via email.

*BioGenx2b*
*Twitch Broadcaster | Tech Aficionado*

*Bad News Gaming Stream TeamTEST Gaming Stream Team*
*AMD Red Team Plus*
*http://twitch.tv/BioGenx2b <http://twitch.tv/BioGenx2b>*

nefarius, 2016-12-03:
So?

BioGenx2b, 2016-12-03:
So I'm not going out of my way to remove my email sig every time I reply
because it's a hassle. I'm reporting an issue to you and giving some
feedback on my recent troubles with your software. Let's just stick to that.

*BioGenx2b*
*Twitch Broadcaster | Tech Aficionado*

*Bad News Gaming Stream TeamTEST Gaming Stream Team*
*AMD Red Team Plus*
*http://twitch.tv/BioGenx2b <http://twitch.tv/BioGenx2b>*

charlesmilette, 2016-12-03:
You can edit your comment afterwards on GitHub and remove the signature.

nefarius, 2016-12-03:
The issue contains a vague statement which I can't replicate so I'm at a loss of what to tell you...

BioGenx2b, 2016-12-03:
I have a Windows 10 AU machine where this replicated through multiple fresh
installs. I install the drivers and services, restart, then they're gone.

• The DSx service can't be restarted but still exists. When trying to start
it, Windows tells me it's "flagged for deletion."
• Reinstalling the VBus service and DSx service resolves the issue, but is
tedious.

Is there a set of command line arguments I can use currently to reinstall
the entire package, or at least the services?

*BioGenx2b*
*Twitch Broadcaster | Tech Aficionado*
*Bad News Gaming Stream Team*
*TEST Gaming Stream Team*
*AMD Red Team Plus*
*http://twitch.tv/BioGenx2b <http://twitch.tv/BioGenx2b>*

nefarius, 2016-12-03:
If the service get's marked for removal "on its own" there's more wrong than just the AU. Maybe your Antivirus is interfering or smth. this simply shouldn't happen.

BioGenx2b, 2016-12-03:
This is a fresh install with no antivirus in the system. Steam and Firefox
are installed, then your driver package.

*BioGenx2b*
*Twitch Broadcaster | Tech Aficionado*
*Bad News Gaming Stream Team*
*TEST Gaming Stream Team*
*AMD Red Team Plus*
*http://twitch.tv/BioGenx2b <http://twitch.tv/BioGenx2b>*

DanHolli123, 2016-12-05:
it's the Anniversary Update, many things broke with that update

closed 2018-01-26 by nefarius